mendax0110/FFRESW #2: Vakuum-SoftwareRequirements

FelixLatzer, 2024-09-13:


mendax0110, 2024-10-11:
Sensoren:
https://uk.banggood.com/CJMCU-388-BMP388-Digital-Temperature-and-Atmospheric-Pressure-Sensor-with-Low-Power-Consumption-24-Bits-Low-Noise-p-1470290.html?cur_warehouse=CN&rmmds=search

https://www.amazon.co.uk/AZDelivery-Generic-Barometric-Pressure-Raspberry/dp/B07D8TPVVY/ref=sr_1_2?dib=eyJ2IjoiMSJ9.MSdUR-pqklLTgFPxWNncZiVJHSSvuzVIM-KvYCuNdhP-ki3JDbAsqQJWa6K8wGmdckIRUa6bU0e_uN14O7bGtayZISePFzneccopl_jK5WRwAf4abJMSSUqeLQB5NvXe-tWy57WSV37OkK7WRTZ_zsIL7eCnJyOH8_eDSe61DDBxlUTVyoHUnYuUZzrrILc0opKKJhSIKWJplx19SwVJL2XnxtWLusz-Ftzd3mTjmMsyu2ZbLlN9GN0hYzD9A398S3N_IMU1-dwWHP7HAVOA8fMC5nEOcD2nN5segktMUHE.k56L8_7WWYCwRGSoK_ApxYgBe0XA9kCKpzqIX3BzkLI&dib_tag=se&keywords=bmp280&qid=1728641577&sr=8-2

https://www.reddit.com/r/arduino/comments/16a5yea/arduino_vacuum_pressure_sensor/

FelixLatzer, 2024-10-18:
@mendax0110 Im GoogleDrive unter "G:\Meine Ablage\Diplomarbeit\HW\uc\Pinbelegung.xlsx" können die benötigten Pins eingetragen werden. Dadurch behalten wir den Überblick welche Pins zur Verfügung stehen.

mendax0110, 2025-01-26:
Wir nutzten den von VAT zur verfügung gestellten uC als slave für den MEGA2560. Die SW-related requirements sind damit festgelegt und wir nutzten die ethernet-schnittstelle inkls. der API von VAT

mendax0110, 2025-02-23:
basic implementation done, set and get parameter working. Next set and get compound 1, 2 and 3. then REST Endpoint creation for set, get and set_compound and get_compound