mendax0110/FFRESW #18: ESW-HAS-Communication

FelixLatzer, 2025-01-18:
Wie soll Komunikation erfolgen?

- Format
- Zeitmanagement (Arduino Zeit setzen)
- verfügbare Endpoints

FelixLatzer, 2025-01-18:
Mit @mendax0110 besprochen:

- Zeit wird von HAS an ESW gesendet (POST auf endpoint, bei erstem verbinden und dann allle 10 minuten oder so)
- Sensordaten werden von ESW gepuffert (nur aktuellster Wert) und immer wieder von HAS abgerufen (GET auf endpoint)
-Für jeden Sensor einen eigenen Endpoint erstellen (http://ESW/temperature, https://ESW/pressure, .....)
-Format für Sensordaten:
{
"sensor_name": "temperature_sensor_1",
"value": 22.5,
"unit": "Celsius",
"timestamp": "2025-01-04T15:30:00Z"
}

mendax0110, 2025-01-26:
Basics wurden implementiert und getestet inkls. python script zum testen der endpoints.
@FelixLatzer wir sollten hier noch das Zeitmangement besprechen

FelixLatzer, 2025-03-07:
Zeitmanagement wurde besprochen:
Der HAS bietet eine API route an, auf welche die ESW zugreifen und durch GET requests die zeit geholt werden kann

closed 2025-03-07 by FelixLatzer