mendax0110/FFRESW #23: Research Dosimeter/Bubbledetector

FelixLatzer, 2025-02-07:
DIY Dosimeter:
https://www.hackster.io/yellowknife/diy-dosimeter-for-the-youngest-engineers-efe785

mendax0110, 2025-04-18:
https://www.mdpi.com/1424-8220/23/1/329

https://indico.cern.ch/event/1044975/contributions/4663817/attachments/2394211/4093387/Slides.pdf

https://eljentechnology.com/products/neutron-detectors/ej-426

https://physicsopenlab.org/2017/01/30/neutron-detector/

https://physicsopenlab.org/wp-content/uploads/2017/02/SlowNeutronCoronaCounterTubes_1v10-1.pdf

https://seansoleyman.com/fusor-neutron-generator/

https://neutrons2.ornl.gov/conf/nxs2013/lecture/pdf/Carpenter-detectors2013.pdf

https://fusor.net/board/viewtopic.php?t=15444
https://fusor.net/board/viewtopic.php?t=14477
https://www.ebay.com/itm/404428469057?_trksid=p2332490.c101875.m1851&itmprp=cksum%3A4044284690571c1dacab6b5f45ad85c1a7127c2e77d0%7Cenc%3AAQAKAAABYHKEKKNMUePryBh1zZl0qrjonWK6%252Bbjh8v6%252FxTRbw1Loxqx736BoO2V7ozv6W4xuOBJjwGhPLWk9lnRB2teM6PFgOrZGVcTwpTth7MW79p066SWQrB1VasW1Di58xl1RMBXsH%252BaNxpzil0t8bdqbPkgJlZgvtHVYNSkxo5xG%252Bq21w2Ndxu0%252B09f9wzuPV730aHIyYiAOEcjFb3mvidMUNjupZqvLpOT4h3Nxa4dBOYCENE1JOMQtKbLqfubYfFGHmwiIFMKNmhwrb7pb3WsZs4u5Mi7UR7PNupDpibWAzpDYp8g356a%252Bh5osKY4Go9%252BCm%252B1IR152xlsWIgVtSaxnBagP1l9%252BD4N%252BEi2XFTLBJIEStkjQzlBduDGoo37z0Siyw2CZSdGUGPiXw0y7tbJLo9IhkUysjcFf2YTU9IZDsvdE41ScL176%252BnXLLPoxPcxI2yD1CRyQ7HxoZx0lVQ%252B3VoQ%253D%7Campid%3APL_CLK%7Cclp%3A2332490&itmmeta=01JS4MZACGMZMC8FN44EQ7CC2K

https://www.fusor.eu/monitors.html#neutron_detector_and_counter
https://fusor.net/board/viewtopic.php?t=5408

mendax0110, 2025-05-17:
Send two mails to AKW in Swiss and the one in Austria to talk about neutron dosimeter.

mendax0110, 2025-06-06:
Also sent two mail to iter.

mendax0110, 2025-07-30:
Basic Implemenation Done, waiting for the meeting and the Hardware then i can do the first few tests.